SmallpTsai/hexapod-v2-7697 #4: Corrupt/unprintable joint_cross.stl

khenderick, 2019-05-14:
The model `join_cross.stl` seems to be broken. A first indication is that pictures of your hexapod and the images in the repo show the corners flat, while my slicer and certain stl viewers display it with rounded corners.

It seems fusion 360 can't automatically repair it, and I couldn't figure out the exact measurements to redraw it.

Any possibility you can re-export it?

jmfinn9, 2019-05-15:
Does this look better? I ran it through makeprintable.com
[fixed_stl_joint_cross.zip](https://github.com/SmallpTsai/hexapod-v2-7697/files/3180500/fixed_stl_joint_cross.zip)

khenderick, 2019-05-15:
Looks and slices perfect. Thanks a lot :smile: . I tried a bunch of online repair services myself, but good to know this one managed to fix the issues.

closed 2019-06-15 by SmallpTsai