SmallpTsai/hexapod-v2-7697 #6: use servo sg90 instead MG92B

pindarogb, 2019-10-28:
You have to check if the sg90 servos fit the dimensions of your project, because I want to have print to use the sg90 motors.

thanks

ArvinRoad, 2020-11-29:
我也有这样的疑问，sg90的伺服可以代替Mg92b吗？

kooliha, 2021-09-02:
+1 The same question 

Can MG90D use instead of MG92B? Size is a bit different and torque for sure
https://www.towerpro.com.tw/product/mg92b/
https://www.towerpro.com.tw/product/mg90d-2/

ValeraDanger, 2022-01-04:
I am using mg90s (analog of sg90s with metal gears). They have same size like mg92b, but hight is smaller. I already made one leg and all right. But you have to use little bit longer pins (about 8mm). The only one problem I see - connection legs to the main part. Body is thicker then legs. But I think, this problem will be solved.

Omarinh, 2022-01-17:
> 

so do you use sg90s instead of mg92b and the hexapod still move
im afraid it's too heavy for that servo

ValeraDanger, 2022-01-17:
> 

No, I'm using mg90s. This is sg90s, but only with metal gears. I have already made only mechanical part. It is rather lightweight (540g with all my servos without battery and electronics). Robot standing without any powering, so there willn't a problem for servos, I think.

Omarinh, 2022-01-17:
540 g without battery and eletronics i think they are the most of the weight but this is a bit reassuringthanks

Omarinh, 2022-01-17:
> 540 g without battery and eletronics i think they are the most of the weight but this is a bit reassuring, thanks for info

ValeraDanger, 2022-01-18:
> > 540 g without battery and eletronics i think they are the most of the weight but this is a bit reassuring, thanks for info

I will use 4*18650. It is about 200g. Esp32 and pwm controllers have no weight in this situation. Robot is weighing less kg. It is  normal for 18 mg90s, IMHO)
But I recommend you use metal servos anyway

Omarinh, 2022-01-18:
Okay thanks, esp32 has more pwm do you use  two pca9685 anyway?

ValeraDanger, 2022-01-18:
> Okay thanks, esp32 has more pwm do you use two pca9685 anyway?

It has 16 pwms, I need 18 :)
Yeap, I'm using 2 pca9685. I think, the code will be quite unreadable, if some servos Wil be on esp32, and some on pca9685

Omarinh, 2022-01-18:
Okay thanks i forgot to see the pinout of the esp32 :)

UlrichKlotzsche, 2022-11-05:
> I am using mg90s (analog of sg90s with metal gears). They have same size like mg92b, but hight is smaller. I already made one leg and all right. But you have to use little bit longer pins (about [...]

I altered the thigh_bottom so that the thigh fits to the body, when you use MG90s servos. Here you go: https://github.com/UlrichKlotzsche/Hexapod-TowerPro_MG90s/blob/main/thigh_bottom_MG90s.stl

As I am new to 3D Printing and CAD, this is not perfect but it works for me.