SmallpTsai/hexapod-v2-7697 #10: My first electronic project.

vertigo475, 2020-07-10:
I would like to thank you for all the detail given. It was my first electronic project and it' a success... 

I've ordered the pcb with your files and a success again. 

[image]

Thanks for sharing, very good quality of files. 

Olivier

carlo5, 2020-07-26:
Hi, I wanted to ask you about a doubt in the assembly, you ordered the PCB already with the parts installed or did you install them. Because I have looked at the entire project, but I cannot find something that indicates the correct order of assembly according to its components . I just found the list of components. Or if somewhere you found BOM file (Bill of Materials) to be assembled by the PCB manufacturer.
I already appreciate the response and sorry for the inconvenience.
A big greeting from Argentina

carlo5, 2020-07-26:
Hi, that's good! It looks very well assembled, I congratulate you! I wanted to ask you about a doubt in the assembly, you ordered the PCB already with the parts installed or I install them because I have looked at the entire project, but I cannot find something that indicates the correct order assembly according to its components. Or if somewhere you found BOM file (Bill of Materials) to be assembled by the PCB manufacturer.
I already appreciate the response and sorry for the inconvenience.
A big greeting from Argentina

carlo5, 2020-07-26:
Hi, that's good! looks very well armed, congratulations! I'm trying to build one, and I wanted to ask you about a question that I have in the assembly, you ordered the PCB already with the parts installed or you installed them because I have looked at the entire project, but I can't find something that indicates the correct order assembly according to its components. Or if somewhere you found BOM file (Bill of Materials) to be assembled by the PCB manufacturer.
I already appreciate the response and sorry for the inconvenience.
A big greeting from Argentina

vertigo475, 2020-07-26:
Hi,
you Will find the BOM under the link below,
https://github.com/SmallpTsai/hexapod-v2-7697/blob/master/electronics/README.md

I had for reference for what to solder where on the files BRM (indications to order pcb boards). I start personnally oy the most little pieces (2 smallest with 2x14 pins) and after the SMD components at least the mini dc, and the 2.54 pins. It Wars my first project so i take a microscope to solder it proprely, with a really small iron and a lot of patience 😀

I don't Know if the manufacturer can assemble it.  I order them without components by first pcb in Hong Kong with good quality and fast delivery and good price. I order 10 pieces of subboard and 5 pieces of main board in case of bullshit solder.

At your disposal for questions, regards from switzerland

Olivier Corbaz

wudikukuqi, 2020-08-16:
Why do I make it according to the routine, but can't drive mg92b to run after starting，Software upload, signal waveform is no problem，Have you ever encountered such a situation

wudikukuqi, 2020-08-19:
> 
> 
> [image]
> 
> I would like to thank you for all the detail given. It was my first electronic project and it' a success... 
> 
> 
> 
> I've ordered the pcb with your files and a success again. 
> 
> 
> 
> [image]
> 
> 
> 
> Thanks for sharing, very good quality of files. 
> 
> 
> 
> Olivier 

Why do I make it according to the routine, but can't drive mg92b to run after starting，Software upload, signal waveform is no problem，Have you ever encountered such a situation

vertigo475, 2020-08-19:
Hi, did you put the jumper in the 2 pins of the main board, just near the development card ? 
Did you have the connexion with the appli for phone ?

wudikukuqi, 2020-08-20:
> Hi, did you put the jumper in the 2 pins of the main board, just near the development card ? 
> 
> Did you have the connexion with the appli for phone ? 

Thank you for your reply. I replaced the jumper on the main board with a switch. Applications can be properly connected to the development board[image]

vertigo475, 2020-08-20:
Excellent ! Enjoy ! 👍🏻

wudikukuqi, 2020-08-20:
> Excellent ! Enjoy ! 👍🏻

I'm sorry I didn't make it clear yesterday, Although the app can connect to the phone,But it still can't drive the MG92B.[image][image]

vertigo475, 2020-08-21:
I remember one of the 2 sub boards have a difference by join 2 points with soldering somewhere like in this pictures, did you make it ?
[image]

vertigo475, 2020-08-21:
I don't remember exactly where, it could be a solution. Sorry but I'm limited in electronics capabilities 😉

as321654987, 2020-09-16:
> > I remember one of the 2 sub boards have a difference by join 2 points with soldering somewhere like in this pictures, did you make it ?
> > [image]
> 
> -Thank you for your support. I'll continue to investigate the reasons

were you able to find the problem? I think I am having the same problem as you did. I was able to run the servos on the initial test run but after calibrations, the servos are not responsive and more.

wudikukuqi, 2020-09-20:
> > > I remember one of the 2 sub boards have a difference by join 2 points with soldering somewhere like in this pictures, did you make it ?
> 
> > > [image]
> 
> > 
> 
> > -Thank you for your support. I'll continue to investigate the reasons
> 
> 
> 
> were you able to find the problem? I think I am having the same problem as you did. I was able to run the servos on the initial test run but after calibrations, the servos are not responsive anymore.

I'm very sorry that I can't help you for the time being, because I haven't found out what's wrong.

Handschrauber, 2020-10-19:
I have the same problem.
During the first test run, the Hexapod stopped after a short time and did not move until it was restarted, and it has not moved at all since the calibration.
The Arduino IDE gives me some errors when compiling, which are probably related to it. Since I am not familiar with programming, I do not know where the problem is and how I can fix it.

Can anybody help me further?
[Arduino Error.txt](https://github.com/SmallpTsai/hexapod-v2-7697/files/5403360/Arduino.Error.txt)

vertigo475, 2020-10-19:
Hi, 
Sorry for your experience. 
I don't had this problem. It was my first arduino project then I spend a lot of time to find the right parameter in arduino ide. But when I found it and the libraries at the right place. It goes straight right without errors in ide and without touching the code. If it could help you I can send screenshots of the parameters ? I don't know if it can help !

Anybody can help Handschrauber for this matter ?  

🤞🏻🤞🏻🤞🏻

Screwmaster37, 2020-10-20:
Thank you for your prompt reply,

I would like to try it with the right parameters.
With which versions of Arduino and LinkIT was the project developed?

I've noticed that "Examples for Linkit 7697" / .LRemote / RemoteControl get similar error messages when compiling.

Could it be that the .LRemote library has changed?

Thanks for the support

vertigo475, 2020-10-20:
Hi, 
It's reason why I bought the "Linkit 7697" it's easier to setup if exactly the same than original.
With the LRemote and Linkit 7697, just push the code in linkit and charge LRemote and all allright.

With different programmation card I' haven't any experience (juste one little tool with arduino nano). Sorry.

Good Luck !

Screwmaster37, 2020-10-21:
I followed the parts list exactly.
The hexapod also worked briefly, but always stopped after a short time.
In the meantime it no longer goes into the home position when it is switched on.
The servos are all supplied with power, but I have not yet been able to detect a PWM signal.

I would like to try the screeshots. Maybe it's just a wrong setting in the Arduino IDE.

Many thanks for the help

vertigo475, 2020-10-21:
With my meaning and understanding, if the hexapod **work** briefly, I think the code input and the parameter with the linkit are not really in question. 

I think if you have the control sceen on your smartphone with all the functionality , it could means that the input in linkit card is ok, because this setup is saved in linkit and not in your mobile phone. I think at electronical problem instead.

Eventually Overcharge in volts and the linkit cut before fatal overcharge? the voltage of 7 miniDC is right to 5-6V ? Maybe a bridge in soldering the 2 microchips (because problem with signal) ?

This is just a guess, if a specialist is reading this, please help Screwmaster37 ;o)
Thank's in advance.

chen-0220, 2020-11-03:
Hello,everyone,IamfromChina.Afterseeingthisvideo,Ireallyhopetomakeitbymyself.However,duetothelimitationofthewebsite,IcannotdownloadthefilesonGithub，my email:1215149771@qq.com

kinghlovo, 2021-02-13:
Hi,do you know what software the Android phone uses to control this? I am a little puzzled seeing so many python programs

vertigo475, 2021-02-13:
@kinghlovo pls find the image below with details. I take the LinkIt remote control. 
[image]

kinghlovo, 2021-02-13:
@vertigo475 Thank you very much, hope I can make it soon

dirk-asam, 2021-02-19:
Hello Vertigo,

do you have another 3 of these boards (1 Main board, 2 sub boards) you had used? The boards I have bought are wrong (some connections are wrong or not there). 
P.S. it‘s also my first electronic board
Best regards Dirk

vertigo475, 2021-02-19:
@dirk-asam, 

hi, yes I have 3 main boards and 6 sub boards left, avaiable if needed.

We can talk about that per mail directely  --> vertigo475@hotmail.com..

Best regards, Olivier

carlosimparato, 2021-06-05:
Buenas tardes. Después de leer todos los comentarios aún no he podido lograr subir el programa a la linkit.

vertigo475, 2021-06-06:
Hi, 
[image]

carlosimparato, 2021-06-06:
es primera vez que trabajo con linkit y no se como pasarlos programas a la placa.
no se conecta a mi pc. linkit remote no reconoce la placa. hay que hacer algo previo?

vertigo475, 2021-06-06:
Olà, 
It was my first time too with LinkIt, I used ide arduino to link with the pc and put the downloaded code into LinkIt. Once the code in, the connection with the app come automatically. To import the libraries in LinkIt. I followed a tutorial find in internet / YouTube. Possible some can help more Carlo in this discussion ? Hope so...🤞🏻

carlosimparato, 2021-06-06:
gracias vertigo475

jyothisvikram, 2021-06-09:
Thank you vertigo475 for all your support above. I am not able to find a place where I can buy Linkit 7697. Any idea where I can buy it?

wudikukuqi, 2021-06-10:
> Thank you vertigo475 for all your support above. I am not able to find a place where I can buy Linkit 7697. Any idea where I can buy it?

ebuy

wudikukuqi, 2021-06-10:
> > Thank you vertigo475 for all your support above. I am not able to find a place where I can buy Linkit 7697. Any idea where I can buy it?
> 
> 

You can buy it on ebuy, but it's super expensive. I bought three without a large-scale price increase.

vertigo475, 2021-06-10:
Here in switzerland we cay buy it for approx 25 us without sending costs www.galaxus.ch. Comes very quickly...

jyothisvikram, 2021-06-10:
Thank you for the update... I can’t find any here who got that! Have any of you used any other Arduino chip for this project?

carlosimparato, 2021-06-12:
no se puede realizar con otra placa porque todo el circuito de las PCB estan realizadas para el 7697,  yop tengo el linkit 7697 y el 7688 pero el 7697 no logro hacer que mi pc lo reconozca. no se porque.

carlosimparato, 2021-06-12:
necesito saber si mi placa de linkit 7967 esta buena. ustedes cuando conectan la placa a la computadora, es reconocida automáticamente? Gracias

vertigo475, 2021-06-12:
It depends with which program do you work with. With ide arduino you have to setup a bit. I don't know how it is withe the others programs. It a long time ago but I found a video in YouTube explaining how to link LinkIt  with ide arduino, I remember it was in English I don't know if it exist in espnol

carlosimparato, 2021-06-15:
el idioma no importa, el problema cuando conecto la linkit a mi computador no hace nada, no suena, no me dice nada, como si el linkit no estuviese conectado. he instalado varios controladores pero ninguno reconoce al linkit. he llegado a pensar que la placa esta dañada.

wudikukuqi, 2021-06-16:
> Thank you for the update... I can’t find any here who got that! Have any of you used any other Arduino chip for this project?

Which country are you in? I have two pieces of surplus here, which can be transferrable to you.

carlosimparato, 2022-01-11:
Hola de nuevo. los programas de python que trae la carpeta de software, hay que cargarlos en la linkit? yo cargue el archivo de arduino perfectamente pero no se que hacer con los archivos de python.   Una ayudita por favor.  mil gracias.

vertigo475, 2022-01-11:
Buenos dias,
gracias por tu mensaje.

No necesitaba los archivos de python. solo arduino

Buenas tardes
olivier

LuardG247, 2024-07-11:
Hello vertigo475,i'm new to this kind of project so hope i could be ask for your help.Can you show me what is the degree of each servo when you assemble it into the leg? Many thank for you support.

vertigo475, 2024-07-11:
[image]

vertigo475, 2024-07-11:
[image]