SmallpTsai/hexapod-v2-7697 #12: 3D-Printed Parts do not fit together

TimKlaas, 2020-11-28:
Hey,

we tried to rebuild the Hexapod as described, but after printing all the parts we noticed that they not fit together properly and when taking a look at the STL files the sizes also dont match up. One example would be connecting the foot_top and foot_tip or the fitting for the motor. What is the reason for that, did we miss something?

MRCamiloLagos, 2020-12-04:
can you help me? i need to get the screw/bolts that are used where i can get in internet? i think the size are 1M ?
I From Honduras i would appreciate i need for a project camiloflores1986@hotmail.com my email

@MRCamiloLagos

rx290, 2024-11-14:
There is this calculator which help you resize the fitting items, here is the link for it, clone it and let me know if this worked for you.
https://docs.google.com/spreadsheets/d/1LO37kJxJx8rW_mv_S8CdKpyp7A_Vs2d3BoclptP_St4/edit?gid=0#gid=0