SmallpTsai/hexapod-v2-7697 #13: screw/bolts sizes

MRCamiloLagos, 2020-12-04:
can you somebody help me?  i need to get the screw/bolts that are used where i can get in internet? i think the size are 1M ?
 I From Honduras i would appreciate i need for a project camiloflores1986@hotmail.com my email

schnello, 2021-01-20:
Take a loot to the BOM: https://github.com/SmallpTsai/hexapod-v2-7697/blob/master/mechanism/BOM.md
All screws are listed

MRCamiloLagos, 2021-01-20:
I thought the hexapod Leg Assembly leg bolts were M1.. 

I look M2  a little big but thanks... i will try with M2

MRCamiloLagos, 2021-01-20:
Thanks My Friend  and this video https://www.youtube.com/watch?v=oaAE5fC09KQ&feature=emb_logo they used M1?
[https://i.ytimg.com/vi/oaAE5fC09KQ/maxresdefault.jpg]<https://www.youtube.com/watch?v=oaAE5fC09KQ&feature=emb_logo>
Hexapod Leg Assembly - YouTube<https://www.youtube.com/watch?v=oaAE5fC09KQ&feature=emb_logo>
Hexapod v2.1 - Leg assembly instructionThis video demostrate how to seemble 1 Leg of Hexapod (total 6 required)For hexpod running video, please check https:/...
www.youtube.com

Camilo?? FLores
97738475
Manten la cabeza en la direccion Correcta, y no tendras que preocuparte por tus Pies..

schnello, 2021-01-30:
I build it with M2 and M2 fits perfect:
[image]

MRCamiloLagos, 2021-02-08:
THANKS MAN last question do you have a link where to get it in Internet amazon o something i want buy from Amazon