SmallpTsai/hexapod-v2-7697 #18: Please change LinkIt 7697. We cant find this :( . Many people like your project!!! But we cant find LinkIt 7697 : ( . And Thanx your work!!!

LhamoJam, 2021-11-17:
Please change LinkIt 7697. We cant find this :( . Many people like your project!!! But we cant find LinkIt 7697 : ( . And Thanx your work!!!

LhamoJam, 2021-11-17:
Some country buy linkit7697 have to wait 3 moth to get Linkit 7697 : (

Tomtom5893, 2021-11-25:
I got one but i cant find the sdk so i cant flash them with the arduino software...

Tomtom5893, 2021-11-25:
https://www.amazon.de/AZDelivery-ESP32-NodeMCU-gratis-eBook/dp/B07Z83MF5W/ref=asc_df_B07Z83MF5W/?tag=googshopde-21&linkCode=df0&hvadid=427788576373&hvpos=&hvnetw=g&hvrand=11084735166114881294&hvpone=&hvptwo=&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9041714&hvtargid=pla-898234157885&psc=1&th=1&psc=1&tag=&ref=&adgrpid=99889209552&hvpone=&hvptwo=&hvadid=427788576373&hvpos=&hvnetw=g&hvrand=11084735166114881294&hvqmt=&hvdev=c&hvdvcmdl=&hvlocint=&hvlocphy=9041714&hvtargid=pla-898234157885

Maybe this one will do it?

it has the same hardware specs (at least thats what i can say)
maybe it needs a little bit of code change but then you are good to go

rookiepeng, 2022-01-01:
I have made it work with Raspberry Pi Zero W. Increased the body width a little, and full rework on the electronic and software.
https://github.com/rookiepeng/hexapod