SmallpTsai/hexapod-v2-7697 #21: I have spare PCB's if someone want.

Tomtom5893, 2021-12-17:
I ordered more PCB's as needed because I didn't know if something was going wrong.
I still have some available if someone wants them.

(Shipped from Germany)

Bachir-Tarif, 2021-12-17:
Hi, im a student i am currently bulding the hexapode if you could do the
shipping i am in France.

Tomtom5893, 2021-12-17:
@Bachir-Tarif can you send me your mail without @ like example(at)something(dot)com ?

Bachir-Tarif, 2021-12-17:
I didn't understand exactly, you mean like this:
***@***.***

Tomtom5893, 2021-12-17:
github censors the emails. if you send me the email the normal way it looks like this example@yahoo.com but if you send it like this example (at) yahoo (dot) com it works

Bachir-Tarif, 2021-12-17:
> looks like this ***@***.*** but if you send it like this example

Aah ok, then it's
bachir.cpu(At)gmail( dot)com

Tomtom5893, 2021-12-17:
> […](#)

I send you Email. did you receive it? if so, you can delete your comment

Lorenzik66, 2021-12-25:
> J'ai commandé plus de PCB au besoin parce que je ne savais pas si quelque chose n'allait pas. J'en ai encore de disponible si quelqu'un en veut.
> 
> (Expédié d'Allemagne)
> 
> J'ai aussi beaucoup d'autres pièces nécessaires comme des résistances, des leds, des cavaliers, etc.

bonjour, je suis intéressé pour fabriquer le mien , comment peut on faire ? je suis en france.
voici mon mail : courrierlorenzo😀gmail(dat)com
merci d’avance et joyeux noël

GGabi0101, 2023-02-21:
Moin könntest du mir auch deine Email wie oben beschrieben schreiben? wäre an einem pcb set interessiert

Tomtom5893, 2023-02-21:
> interessiert

tomtom5893 [at] Gmail.com